BCN3D/BCN3D-Moveo #2: Missing STL files: T2M1D, T2M1L, T3M1, and T4M

jakegaecke, 2016-08-01:
In the User Manual pdf, steps 4, 5 for Module 2 Assembly (page 11) step 2 for Module 3 Assembly (page 20), and step 2 for Module 4 assembly (page 22), all reference parts that don't seem to be published: T2M1D, T2M1L, T3M1, and T4M. These parts are also listed in the BOM. Can these parts be published?

jamesarm97, 2016-08-01:
I believe those parts are in the source files. I had the same problem but was able to generate them from the solid works files. I am attaching the ones generated from the CAD files. I guess someone can add them to the github as I haven't checked them out.

[Txx_Parts.zip](https://github.com/BCN3D/BCN3D-Moveo/files/394063/Txx_Parts.zip)

airshares, 2016-08-01:
I am looking for the .stl of those files as well

jamesarm97, 2016-08-01:
Check my previous comment. It should have a zip file attached with what I think are those files if someone can verify.

airshares, 2016-08-01:
Yep I see them now.  Thank you!  I did not look at that file earlier.

kitusmark, 2016-08-02:
@jamesarm97 if you could make a pull request it would be awesome as you have already generated the files.

jamesarm97, 2016-08-02:
I did the Pull request, someone will need to confirm and close it.

jakegaecke, 2016-08-04:
Exactly what I needed. Thank you!

closed 2016-08-04 by jakegaecke