BCN3D/BCN3D-Moveo #3: Could someone please provide more accurate information on the motors

IAmOrion, 2016-08-02:
Could someone please provide more accurate information on the motors, I'm trying to source parts in the UK, and the motors listed in the BOM are too generic.  The only 1 with enough info was the Nema 14 36mm SM35HT36 > that I can find, but not a clue on any of the others.

miguelrochajr, 2016-08-02:
I was able to find all the parts at www.aliexpress.com typing the specifications given on the Bill Of Materials. Some of the Nema's have some little differences such as the protruding connection at the sides like [this one](http://www.aliexpress.com/item/Nema-17-Stepper-Motor-bipolar-4-leads-34mm-12V-0-4A-26Ncm-36-8oz-in-3D/32376850840.html?spm=2114.01010208.3.12.nZxJHy&ws_ab_test=searchweb201556_0,searchweb201602_2_10057_10056_10055_10037_10049_301_10059_10033_10058_10032_10017_405_404_10060_10061_10062_412,searchweb201603_6&btsid=ad284de6-6f86-483b-85b6-3c3575afee5e). For the rest, I just looked at thee Datasheet.

IAmOrion, 2016-08-02:
Thanks miguelrochajr, that's good to know, I'll take a look at AliExpress now.

Have you completed your build?  If you get a chance could you paste links for the motors from your order history?

Kind regards

marcovanschagen, 2016-08-07:
Same question, and I will purchase trough ebay.

marcgenevat, 2016-08-07:
Have you estimated the approximated costs of the hardware?  I'm the CEO of a edtech company and I can't wait to see how my students react with this awesome project. Thank you guys!

kitusmark, 2016-10-03:
Hi people! 

Take a look at the updated BOM file as we have added some extra info and corrected other. 
https://github.com/BCN3D/BCN3D-Moveo/commit/0866a92501277636f76000a195d8a16d44b5b476
Thanks for the feedback!

@marcgenevat the cost of hardware is arround 300-400€ depending on where you buy the parts. Everything is pretty standard and you'll be able to source equivalent parts.

AlexanderZhang11, 2016-10-10:
hi IAmOrion ,have you finished your work? Do you know how to connect wiring diagram?

helgrind, 2017-07-04:
I just bought these. Missing the first motor though as I hear that one is a contentious choice so we'll buy it later I think. Prices are in pounds sterling.

BOM ID  Price   Link

8            23.14  https://www.aliexpress.com/item/Best-Price-Extruder-Gear-Stepper-Motor-Ratio-5-1-Planetary-Gearbox-Nema-17-Step-Motor-OSM/32801086766.html?spm=2114.13010608.0.0.7BMv9q

9            62.61  https://www.aliexpress.com/item/P-romotion-1-PC-Stepper-Motor-Nema23-425oz-in-112mm-3A-23HS9430-CNC-kit/32820939354.html?spm=2114.13010608.0.0.7BMv9q

11          5.97    https://www.aliexpress.com/item/Free-shipping-and-Quality-17HS3401-4-lead-Nema-17-Stepper-Motor-42-motor-42BYGH-1-3A/32723884250.html?spm=2114.13010608.0.0.7BMv9q

3            34.51  https://www.aliexpress.com/item/TB6600-stepper-motor-Driver-Controller-4A-9-42V-TTL-16-Micro-Step-CNC-1-Axis-NEW/32629936087.html?spm=2114.13010608.0.0.7BMv9q

10          14.03  https://www.aliexpress.com/item/1-8degree-35mm-1-4Kg-cm-1A-2phase-hybrid-stepper-motor-NEMA14-Robot-Motor-35HY34-1004/32576815079.html?spm=2114.13010608.0.0.M7L5GG