BCN3D/BCN3D-Moveo #6: Wiring Diamgram is missed

ildarjaleh, 2016-08-16:
Please add wiring diagram to the instruction.

AlexanderZhang11, 2016-10-10:
Yes, I am also puzzled by the wiring diagram problem. I really hope it can be updated as soon as possible.

Mocsany, 2016-10-13:
Yes, please provide wiring diagram!

knoxxr, 2017-09-09:
Need wiring diagram too.
I have ask about power supply.
Moveo use 24v power supply in BOM and use voltage converter 24v to 12v.
Why use 24v power instead 12v power.
Ramps 1.4 normaly use 12v power supply.
Is passible 12V, 320W power supply directly?

xy23, 2018-08-31:
Hello I am close to finish the printing part of Moveo and I will start the wiring process in the next few Days. 
My Question is, is ther now any wiring Diagramm here on Github or can someon give me some hint where I can find one?

fc-lapac, 2019-11-05:
Any news on that?

michelgokan, 2021-07-03:
Where is the wiring diagram?