BCN3D/BCN3D-Moveo #7: Detailed BOM

Warcorer, 2016-08-31:
Hello,

The Bill of Materials isn't very detailed on specifications of parts. Could the list please be updated? Issues such as Power Supply 24V a 12V? Is that supposed to mean 24V DC converter to 12V DC? Also stepper nema 17 R 1:5? Is this supposed to mean a Nema 17 Stepper Motor with 5:1 planetary gearbox? Please review the BOM and update the details so it is easier for other people to make this open source project from BCN3D.

Thanks,

Michael

Lostinthe907, 2016-09-29:
I Am having the same Problems I cannot seem to find the Right parts based on the Description of BOM. common part numbers of users who already built it would be Awesome. I don't know much about this stuff But I want to Learn the most I can. e.g. Belt Numbers, common motor numbers, and maybe places to buy the parts. I am in the US so any help would be Really Grateful!!!!

ShopMonkey, 2016-09-30:
For the belts I went to [PolyBelt](http://shop.polybelt.com/main.sc)

Specifically Here are the:
[The Belting ](http://shop.polybelt.com/16T5-Open-End-Belt-Roll-Polyurethane-with-Steel-Cords-B16T5-MPS.htm)
[The 10 tooth Pulley ](http://shop.polybelt.com/27T5-10-2-Aluminum-10-Tooth-Pulley-1505mm-OD-P27T5-10-2A.htm)
[The 14 tooth Pulley](http://shop.polybelt.com/27T5-14-2-Aluminum-14-Tooth-Pulley-2145mm-OD-P27T5-14-2A.htm)

Keep in mind that these pulleys are not predrilled. Witch lets you match to the steppers if you cant find exactly 8mm shafts. You can drill and ream them on a drill press. Use a good reamer.

Stepper motors I got on ebay.

All the fasteners I got at [McMaster-Carr](http://www.mcmaster.com/#)

The joint axles I got ebay

The rest is easily available .

Open builds has power supplies and other bits.

Lostinthe907, 2016-09-30:
Cool. Thanks! @ShopMonkey

kitusmark, 2016-10-03:
Check out the new BOM file 
https://github.com/BCN3D/BCN3D-Moveo/commit/0866a92501277636f76000a195d8a16d44b5b476
Thanks for the feedback!

volo63, 2016-11-25:
Hi,
got the same problem.
I´m located in Germany and I want to start this project for education.
It will take me a lot of time to "redesign" the project because I´m missing the technical details for some BOM parts.
Is there anyone who did or is doing the project in Europe.
My prefered supplier is Conrad electronic and Reichelt electronic.

Is there anybody out there? :))

Volker

AnykeyNL, 2016-11-25:
HI,

I am based in netherlands and building one. I have almost everything
printed out and sourced all components ( I hope). Starting the assembly
next week probably :-)

Regards,
Richard

avanrijswijck, 2016-12-19:
We are building the moveo @ the kaasfabriek. Education will be the main purpose. When ready and tested we will use it for school projects. We will provide all the parts and start an educational programm for primary schools and high schools.

volo63, 2016-12-19:
Hi, we will also build up the moveo robot arm. Would be nice to work together in a virtuell team. We will use it for our Volkswagen Campus Digital in Hannover to train students.

dutchrazor, 2017-01-01:
Here are some links I found for the most matching stepper motors. (the 34mm NEMA 17 is a 0.9* version, 1.8* was sold out)

STEPPER NEMA 17 60mm  (SM42HT47-1684 1,8o step, 1,6 A, 3600 g·cm)	1	N	Y		http://eu.stepperonline.com/nema-17-60ncm85ozin-064a-bipolar-stepper-motor-17hs240644s-p-19.html

Power Supply 24V 320W	1		Y		http://eu.stepperonline.com/switching-power-supply-350w-24v-146a-for-cnc-router-kits-115v230v-s35024-p-177.html
SERVO MOTOR 180§ 55G TORQUE=13KG/CM	1		Y		http://www.tinytronics.nl/shop/Futaba-S3003-Servo
Stepper nema 17 R 1:5 (17HS19-1684S-PG5 ( step 0,35o , 2Nm, 1,68A))	1	N	Y		http://eu.stepperonline.com/gear-ratio-51-planetary-gearbox-high-torque-nema-17-stepper-17hs191684spg5-p-40.html
Nema 23, 112mm, Ø8mm flat shaft (SM57HT112-3004A step 1,8o 3A 28Kg·cm)	2	N	Y		http://eu.stepperonline.com/high-torque-nema-23-cnc-stepper-motor-113mm-3nm425ozin-23hs454204s-p-127.html
 Nema 14 36mm (SM35HT36-1004A, 1,8o step, 1A, 1400g·cm)	1	N	Y		http://eu.stepperonline.com/nema-14-bipolar-stepper-54v-08a-18ncm255ozin-14hs130804s-p-93.html
Nema 17 34mm (SM42HT33-1334, 1,8o step, 1,33A, 2200 g·cm)	1	Y	Y		http://eu.stepperonline.com/nema-17-stepper-motor-34mm-12v-04a-26ncm37ozin-17hs130404s-p-166.html

bullestock, 2017-03-20:
I have to join the chorus here: The current BOM is a bit of a joke.

First, why on earth have you chosen to make a PDF of a screen dump of an Excel sheet?! Come on, guys! Open the Excel sheet in LibreOffice, then export it to an open format!

Secondly, specifications such as "Chrome steel smooth bar-134 mm" are pretty useless. How can I know which diameter is required?

chexov, 2017-03-20:
You guys know you can make pull request and fix it instead of complaining right? This is how OSS works pretty much.

bullestock, 2017-03-20:
@chexov How can I do that when a) no source exists for the BOM b) I have no idea what the specifications are for those "smooth steel bars"?

bullestock, 2017-03-21:
@AnykeyNL do you have any details on items such as steel bars etc?

bullestock, 2017-03-21:
Digging around in the forum, it turns out that the steel bars should be 8 mm.

volo63, 2017-03-21:
Hi Torsten,

 

we already worked our own BOM.

I don´t have any experience with github.

If you are interested in our BOM I could send it.

We also made the first  -bad - expieriences with the cheap motor drivers.

Also we have some not solved issiues.

What is your robot arm status?

 

greetings

volker

 

 

Gesendet: Dienstag, 21. März 2017 um 13:04 Uhr

volo63, 2017-03-21:
one more:)

 

our cylinders are made from us.

 

if you need I could ask in the workshop to produce more.

Do you need the cylinders?

 

 

Gesendet: Dienstag, 21. März 2017 um 13:04 Uhr

bullestock, 2017-03-22:
@volo63 I found a BOM in Excel format in the forum, but I would like to see yours too. I plan to produce an improved BOM and upload it to github. (I have not yet started on my arm, I am just doing research ATM).

I am not sure I understand what you mean by "cylinders"?

anyrules, 2017-03-23:
Hello All,

I have started to print out the components and meantime I looking for components and Fasteners.
Unfortunately it's really do not easy to recognize all components based on the PDF.
(Some of them withdrawned from the market)
First we need to print out after we can concluded for the  parts. 
If somebody have a fresh detailed BOM and send to me I really appreciate it.
Thanks in advance,
Regards
  Istvan

volo63, 2017-03-23:
Hi Istvan,
We will send our updated list tomorrow.
Sorry, saw your email too late.
Greetings 
Volker 
-- 
Diese Nachricht wurde von meinem Android Mobiltelefon mit WEB.DE Mail gesendet.

anyrules, 2017-03-24:
Hi Volker
not a late. The printings take time.  Sould I send to you my mail address or you can send it in here?
This github is new for me. 
Thanks for your support!
Regrads,
  Istvan

volo63, 2017-03-24:
Hi Istvan,
Yes, send email 
Vloebe@web.de 
Greetings 
-- 
Diese Nachricht wurde von meinem Android Mobiltelefon mit WEB.DE Mail gesendet.

anyrules, 2017-03-27:
Hi Volker,

did you got my mail?
Regards,
  Istvan

bullestock, 2017-03-27:
@anyrules can you please send to me too, at torsten@bullestock.net - thanks

modulab, 2017-03-27:
Hey guys! Can't you just upload it on a drive somewhere and share the link? 
Thanks

Warcorer, 2017-03-28:
Could I have it emailed as well

Mjjagosz@gmail.com

Thanks,

Michael J Jagosz

volo63, 2017-03-29:
Hi,

please send me your email, than I will send my BOM

 

greetings

volker

 

 

Gesendet: Mittwoch, 22. März 2017 um 11:14 Uhr

CaptainMack, 2017-05-02:
If you could send the BOM to c.laursen@montem.io I would be grateful!

helgrind, 2017-07-04:
Does anyone have more information about the M4 and M3 brass inserts? Are there recommended lengths etc? Also, are the spacers aluminium ones? Ta

Aerrios, 2017-08-10:
Hi, 
can you please send me your detailed BOM to travnicek@allcomp.cz

Thank you very much!

helgrind, 2017-08-10:
Can we get the detailed BoM in a forked repository or something publicly available online? It seems bad practice to turn this issue thread into an email request page...

tejas4patel, 2017-08-10:
Thank you @helgrind, totally agree..

dylandesouclin, 2018-01-14:
Bonjour, 
c'est vrai que cette BOM est très compliquée à détailler, je voudrais si possible une BOM plus détaillée 
Merci

joealthaus, 2018-01-18:
https://github.com/BradyHooverDesigns/BCN3D-MOVEO-BOM

russelbradley, 2018-09-04:
I'm confused for the smooth rods.
BOM tells me to get lengths of 42mm, 80mm, 114mm and 134mm. While the solidworks file says 50mm, 80mm, 121mm and 140mm respectively. Which one should I get?

Thanks in advance

mcprakash, 2018-10-09:
Also for the chrome steel smooth bar, what is the sectional dimension ? M8?

russelbradley, 2018-10-09:
> Also for the chrome steel smooth bar, what is the sectional dimension ? M8?

Yes 8 mm diameter.

mcprakash, 2018-10-11:
Just a quick question. We are using a 24V to 12V power converter and a 24V power supply. Any reason why we are not using a 12V power supply directly. Also, which components work on 24V and which on 12 V ? Thanks again.

mcprakash, 2018-10-28:
Had another quick question. Which material would be great for 3D Printing the outer body for long lasting (due to regular wear and tear) and stability of the arm

TacticalPanda028, 2019-04-27:
> @russelbradley还有另一个快速的问题。哪种材料可以很好地进行3D打印，使外部机身持久（由于经常磨损）和手臂的稳定性

Recently I have just started this project, I do not know if your project has been successful? This response may not be timely, and I can tell you that I am using components printed on PLA materials

russelbradley, 2019-04-27:
> @russelbradley Had another quick question. Which material would be great for 3D Printing the outer body for long lasting (due to regular wear and tear) and stability of the arm

I used PETG filaments from esun.

mcprakash, 2019-04-27:
@russelbradley I tried ABS. It is just a mess printing with ABS due to several reasons. There are still a few parts I need to print and then finally assemble them. An assemble video would be great to start, but if you are also building it, it would be great to stay connected and build together.

fc-lapac, 2019-11-05:
The BOM uses a power supply of 24v 15A, and a converter from 24v to 12v, usually people use with 10A. So could I just use a power supply of 12v 30A from head start and forget about the converter? will I use the 24v for anything? I haven't found a good electronics diagram that explain why should i have the 24v

jrgilman, 2021-03-19:
> 
> 
> I'm confused for the smooth rods.
> BOM tells me to get lengths of 42mm, 80mm, 114mm and 134mm. While the solidworks file says 50mm, 80mm, 121mm and 140mm respectively. Which one should I get?
> 
> Thanks in advance

Did you ever figure out the right lengths? Also, the BOM doesn't list a 42mm anymore, and looking at the instructions I can't see one being used anywhere. What was this used for in the past?

johnmbiker, 2022-12-01:
I have printed the parts but not ordered the electronics and bolts yet.  After looking around much, I found this guy's detailed BOM on instructables with links.  I can't vouch for it yet but hope this helps someone.  Look for the google docs spreadsheet download.
https://www.instructables.com/Build-a-Giant-3D-Printed-Robot-Arm/