BCN3D/BCN3D-Moveo #9: Igus rails in BoM -- are these needed?

dtmille2, 2016-09-21:
The following parts are listed in the BoM's but not shown in the instruction manual or the CAD assembly:

LINEAL GUIDES IGUS-WSQ_06-134mm
LINEAL GUIDES IGUS-WSQ_06-114mm
LINEAL GUIDES IGUS-WSQ_06-080mm

Am I missing something or are these not needed?

Thanks!!

ShopMonkey, 2016-09-21:
I'm using regular 8mm rod (shafting) I don't see the need for lineal guides. They are in fact joint axles. Don't skimp on the bearings. I bought some cheep ones and they are unusable.

kitusmark, 2016-10-03:
Hi @dtmille2 and @ShopMonkey 

We've updated the BOM file. Please take a look and get back to me.

https://github.com/BCN3D/BCN3D-Moveo/commit/0866a92501277636f76000a195d8a16d44b5b476

Thanks for the feedback!

dtmille2, 2016-10-23:
Thanks for updating the BoM, the additional stepper motor part numbers are great.

Unfortunately items 12, 13, and 14 are still not fully defined -- what diameter should the bars be?  6mm?

closed 2016-11-20 by dtmille2