BCN3D/BCN3D-Moveo #10: Lineas Guides

BigSus, 2016-09-28:
Estamos preparando lo necesario para montar un brazo Moveo y según el BOM las barras lisas que usáis son de Igus. Concretamente las referenciais como  IGUS-WSQ_06, pero dicha referencia por lo que veo se corresponde con guías (http://www.igus.es/Search?q=WSQ-06), no con varillas. ¿Podéis especificar un poco más que tipo de varillas usáis y su diámetro?

Gracias

kitusmark, 2016-10-03:
Hola @BigSus. 

Hemos actualizado el BOM. Échale un ojo a ver si hemos resuelto tus dudas. 

https://github.com/BCN3D/BCN3D-Moveo/commit/0866a92501277636f76000a195d8a16d44b5b476

Gracias por el feedback

BigSus, 2016-10-03:
¿Faltaría indicar que las barras son de 8 mm no?

dtmille2, 2016-10-23:
Yes what diameter are the bars meant to be?

dtmille2, 2016-10-23:
Just pulled up the SolidWorks assembly file and from the name of the shaft parts it looks like they are meant to 8 mm diameter--please update the BoM, thanks!

[image]