BCN3D/BCN3D-Moveo #11: CAD files don't assemble in Fusion 360

Mocsany, 2016-09-30:
Hello, everyone I am having a hard time to put all the files into one assembly in Fusion 360. Is there anybody out there who has managed to do it? I am new to Fusion 360 and CAD in general so I may be not doing it right. No matter how I upload the files they don't assemble into one whole piece as I would like them to be.

macinblack, 2017-02-02:
It's a Solidworks design...

majoon, 2017-02-02:
There is absolutely no problem with the assembly in fusion 360 as long as all the files are uploaded together and the correct file is selected as the main assembly (there are many sw assemblies in the folder.)

spad, 2018-03-28:
Upload all solidworks files and select "BCN3D Moveo assembly" as top level assembly. It works for me.

soswow, 2018-04-18:
There are still some problems after importing and choosing "BCN3D Moveo assembly" as a main assembly. 
1) All the bolts and blue caps are out of components where the suppose to be, and after enabling history copy-pasting them into appropriate component causes their move out of place
2) There are some empty components with name "Sistema rodaments". Also there is a error document generated.

Henke1983, 2020-02-29:
Here is complete assembly. 
https://github.com/magnuspyke/BCN3D-Moveo/tree/master/CAD

I think this issue can be closed.

soswow, 2020-02-29:
Great! thank you! I can't close it, but it can be.