BCN3D/BCN3D-Moveo #14: A new software to control moveo

darenwai, 2016-10-29:
Finally a software and firmware to control the zortrax or moveo robotic arm.
14days trial. US25 per software. Please support me. Thanks.

Demo: https://youtu.be/wO6RP1ZqzVk

Features:
-Can control two arduino mega with ramps1.4. A total of 12axis.
-12axis can be played in a single gcode file.
-Why 12 axis? Normally a 6axis robotic arm is not sufficient to solve real world problem.
 So, the extended axis are used to assist the robotic arm to perform task.
-Use joystick to control.
-Support two types of gripper. Magnetic and servo.
-Auto recording of gcode.
-Can play line by line of gcode and delete a single line of gcode.
-Can save and playback unlimited gcode file.

Download the software and firmware at https://drive.google.com/file/d/0BzSicdTSN7f9V25XYm5UdkxEMzg/view?usp=sharing
Email: testdebug@hotmail.com

Hassan3ysaaopp, 2023-06-07:
How can i buy this software ?

darenwai, 2023-06-08:
To get the license code for basic edition without computer vision feature:

1.https://www.paypal.com
2.After login
3. click on Tool/Send money
4. Click on Pay for goods or services.
5 Type ***@***.***
6. Click next. You should see Quantum Software
7. Put USD 35
8. On Note. You can put your device ID number on the software registration page.
9. I will send you the serial number to register. Thank you.