BCN3D/BCN3D-Moveo #15: Moveo D-H Parameters

MarcoAntonini, 2017-03-19:
Moveo is awesome project! Thanks for sharing it!
I finished assembly my Moveo, I would like to drive it with a direct and inverse kinematics, someone has already calculated the d-h parameters (Denavit–Hartenberg) of this robot?

MarcoAntonini, 2017-03-27:
Ok, calculated it .
Added a joint-6 to expand sixth dof rotary .

Moveo D-H Parameters (5/6DOF)

MarcoAntonini, 2017-03-27:
Ok, calculated it .
Added a joint-6 to expand sixth rotary dof.

**Moveo D-H Parameters (5/6DOF)**

```
#            L2
#   j2  *----------*   j3
#       |       L3 |
#   L1  |          *   j4
#       |       L4 |
#   j1  *          *   j5
#               L5 |
#                  *   j6
#                 | |
```

joint | Alpha | A | D  
-----|-------|--|--|
1 | 0 | 0 | L1
2 | 90 | 0 | 0
3 | 0 | L2 | 0
4 | 90 | 0 | L3
5 | -90 | 0 | 0
6 | 90 | 0 | L4

kitusmark, 2017-03-30:
Wow @MarcoAntonini !

Awesome job. I would be very interested if you share how you control your Moveo. Is it ROS or something? That looks pretty neat

MarcoAntonini, 2017-03-30:
Yes, LinuxCNC has realtime kernel !
Certainly I will share documentation as soon as possible
I'm studying for stepper signals generation, for now just a simulation but very impressive :

https://www.youtube.com/watch?v=55SpbKL9n8U

kitusmark, 2017-03-30:
That's really impressive. Didn't know about linuxCNC (used Mach3 instead). It seems like a very configurable software as you can render the full robot in 3D. 
Is there a way of generating the trajectories within the software? or any way of make the robot learn the movements?
truly impressive

MarcoAntonini, 2017-03-30:
LinuxCNC is similar to Mach3,It has a python scripting language to create your own robot/machine.
It generate the trajectories in non-linear movements from XYZABC cartesian coordinates using direct kinematics.
For current configuration (5 joint) only X-Y-Z-A-B are supported. Add sixth joint for C Axes.

This simulation using the A,B axis , rotate a effector around a fixed point : https://www.youtube.com/watch?v=SHPSGXtctZM ,seems to properly work :)

pkmcnc, 2017-03-30:
@MarcoAntonini this is great! How did you make a visualisation?

MarcoAntonini, 2017-03-30:
@pkmcnc , use the linuxcnc vismach http://linuxcnc.org/docs/html/gui/vismach.html

abdelaziz94, 2017-05-28:
hello sir did you calculate the inverse kinematics of this robotic arm  ? 
if you did it can you provide it to me please ?

tsibos22, 2018-01-06:
Hallo everyone.
Has anyone the inverse kinematics of moveo in excel?
Thanks

tegri, 2018-11-28:
Any news on this one guys? I have found some inverse kinematic solutions on the Internet. Thought you guys might be interested. Here is link: http://www.chew-z.it/wiki/doku.php?id=moveo:moveo_kinematic

Thing is it is not in English.

EBMare, 2021-04-19:
@MarcoAntonini do you have the DH-parameters now? Cuz I need them too

EBMare, 2021-04-20:
https://stackoverflow.com/questions/67159164/denavit-hartenberg-6dof-moveo-inverse-kinematic-robot-arm?noredirect=1#comment118726837_67159164