BCN3D/BCN3D-Moveo #21: Can now use Moveo with ROS and Moveit

jesseweisberg, 2017-11-24:
I've been working to control the Moveo with ROS both in simulation and the real, physical robot-- so far I've gotten some nice functionality working and wanted to share with everyone.  So far, the following functionality works and you can grab all my work thus far from this repo here, and get it working for yourself:
https://github.com/jesseweisberg/moveo_ros

1) Simulate motion planning for the Moveo in RVIZ using Moveit (i.e. plan and execute trajectories)
[image]

2) Read current pose & position of the simulated Moveo end-effector in 3D coordinate space.  You can also plan & execute a trajectory by inputting a 3D coordinate and pose for the end-effector. 

3) Execute these trajectories on the real robot (using an Arduino MEGA 2560, RAMPS 1.4) so that they match the movement in simulation.

4) There is a ton more to do using ROS to allow the Moveo to do incredible things.  I'm currently working on integrating closed-loop control with additional sensors, along with implementing vision feedback to make the control more robust and see how far we can take the Moveo!  If anyone wants to contribute or has any ideas for the future of this project, please let me know!

**I'll have a video of all this stuff up and working in the next couple of days

I tried to detail all the steps to get it running on my github repo, but probably missed some, so please let me know if you have trouble getting it up and running.  Tested on Ubuntu 16.04, and you may have to make slight modifications if running something else-- but definitely worth it.

-Jesse

alequeiroz, 2017-12-09:
Hello @jesseweisberg i'm very interested in visual recognition with moveo, do you already have a repo that we can contribute?

jesseweisberg, 2017-12-11:
Hi Alessandro, I'll actually have something up either tomorrow or Tuesday that incorporates real-time object recognition with moveo control.  I'll post a link here when it's up.

alequeiroz, 2017-12-22:
@jesseweisberg Any news?

jesseweisberg, 2017-12-24:
@alequeiroz sorry for the delay, but I've got something for you now! 
I added to the existing **moveo_ros** repo so that you can now use real-time object recognition from a monocular image (ordinary webcam) to perform predefined 'pick and place' trajectories that are specific to the detected object.  Feel free to fork and experiment!
[**code & details**](https://github.com/jesseweisberg/moveo_ros/tree/master/moveo_moveit/scripts)
[**video demo**](https://youtu.be/kkUbyFa2MWc)

liz-miller, 2018-04-23:
@jesseweisberg very cool addition to this project. I'm thinking about starting this project soon & can't wait to interface it with ROS. Random question - roughly how long does it take to print the moveo parts? TIA!

mfkenson, 2018-05-04:
I finally decided to source all parts because you guys make it compatiable with ROS! Thanks for integrating the ROS.

I simply buy the printed parts because I dont have a 3d printer.

jcheng09, 2022-02-21:
Hi Jesse, nice work so far with on the project. I'm following through your simulation workflow now. I'm wondering if you had any luck with implementing the closed-loop feedback control. I'm interested in contributing to its development