BCN3D/BCN3D-Moveo #30: Adding Limit Switches to Moveo

Coppertop12, 2019-03-28:
Hello, can someone who has added Limit Switches to their Moveo project please direct me to their work or get in touch? If this is not possible or practical, can someone please explain why?

agneyjadav, 2022-11-09:
Heya coppertop12 if anyone helped you with idea of limit switch, kindly guide us towards it, contact me at agney2001@gmail.com

agneyjadav, 2022-11-09:
> Hello, can someone who has added Limit Switches to their Moveo project please direct me to their work or get in touch? If this is not possible or practical, can someone please explain why?

techsavvyomi, 2024-05-31:
> Hello, can someone who has added Limit Switches to their Moveo project please direct me to their work or get in touch? If this is not possible or practical, can someone please explain why?

Hi, I'm also working on the same arm. Did you find solution to add limit switch to arm ? Thanx in advance.

Coppertop12, 2024-05-31:
Hey! I do not have the files related to this any longer. We did add limit switches and a "homing" program on start up. It was not a difficult modification. 

The arm we made, while functional, had several structural issues. The 3D printed parts were too heavy for the motors, even when we sized motors up. Tensioning the belts so they would not slip cracked the 3D prints. We had a different tool attachment however which was much heavier. 

Good luck!

[image]