BCN3D/BCN3D-Moveo #37: Printing the parts

michaelhotker, 2020-06-06:
do any of the parts require supports at all? I could find anywhere in the manual that said anything about it.

AlfonsoCalero, 2020-06-06:
I don't think you will find an instruction manual at such a low level. There are many parts of the project that will go on your own, although you can always count on people who have carried out the project and will lend you a hand. Regarding the subject of supports, you are supposed to have knowledge about whether an object should use supports or not when printing it.

DominicM1213, 2021-12-12:
Has anyone figured out how to print the 1M1 part? I can't seem to get this lip to print properly.

maedrone, 2021-12-14:
Yo lo. He impreso con soportes y me ha quedado muy bien, sinceramente no utilice ninguna configuración modificada, solo agregue soportes

tkircher, 2022-11-05:
It's clear that whoever designed these parts doesn't understand design for 3D printing. Essentially all of these parts require complex supports, when a few small changes would make them require no supports. (For example, tapered hex nut seats instead of flat ones, which is 3D printing design 101)