BCN3D/BCN3D-Moveo #42: Stepper Motors

robertall33, 2020-10-27:
Hi there - I am really sorry for being a pain in asking silly questions. I fully understand that this project is about 3-4 years old now, and i think it is the most amazing project to complete ever, and thank you so much for supplying so much information without charge, and letting it be open source.

I am having trouble finding the correct stepper motors. Can someone please point me in the correct direction?

Many thanks, Rob

EBMare, 2021-04-07:
You can find all of the Stepper Motors except the Nema 23.
Use this one instead "Nema 23 stepper 3Nm 4,2A 10mm shaft"

robertall33, 2021-04-07:
Hi there - Its all resolved now, and the robotic arm is almost fully assembled with the exception of sorting out the wiring - Thank you

PROPSMITH, 2021-04-15:
Hi man, I want to know who you are?

EBMare, 2021-04-15:
@robertall33 I could help you if you want.

Paul2665, 2026-11-04:
Hello, how did you manage calibrating the Motors? Since a Steppermotor only knows which angle it turned, you need to calibrate its direction in the beginning. i thougth about homing (putting a switch at the endposition of each axis). Do you have any recommendations?