BCN3D/BCN3D-Moveo #46: add source files to repo

moedn, 2021-09-21:
First, thanks for making this hardware open source!

I found lots of export files in this repo (PDF, STL etc.). While those are nice for replication, it's quite hard to edit them for modification or further development.

I suggest the following:

- [ ] add source files in native (or at least editable) format to the repo (e.g. `BCN3D Moveo BOM.xlsx` (or even better: as CSV))
  - so others can contribute
- [ ] remove export files (e.g. `BCN3D Moveo BOM.pdf`)
  - git can't handle binaries in a reasonable way; export formats should be either added to a release or created automatically using CI jobs. creating a release and adding those files there would be the easier option, I guess :)

allyouaskfor, 2021-10-09:
The source files are available as SolidWorks files in the CAD directory. 

The file you mentioned, BCN3D Moveo BOM.pdf, is a document known as a 'Bill Of Materials' which contains a list of parts required to assemble the end-product.

moedn, 2022-06-23:
whoops sorry for that! seems like this was never an issue than, thanks for the response!

regarding the BoM, I'd still suggest to use an editable format rather than PDF, but that's rather a detail

closed 2022-06-23 by moedn